olegtarasov/esphome-opentherm #6: Errors

Alfagek, 2024-08-31:
Hello Oleg,

I did see on ESPHome that you are hard working to get the Openthem in to ESPHome. Keep up the good work!!

I did normally used the settings from @freebear-NC, but i do have the same problem what i have if i use your version.

When i flash the D1 with the DIYLESS MasterOpenThem board, it comes online, but it spitting a lot of the same error.
Is it because you are busy with the files, that it trows this error?

Thanks in advanced!

Error is the logs:
`[15:08:40]type: READ_DATA; id: 3; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[15:08:40][W][opentherm:406]: Protocol error occured while receiving response: type: NO_CHANGE_TOO_LONG; data: 00000000; clock: 1; capture: 00000000000000000000000111111111; bit_pos: 0
[15:08:40][D][opentherm:370]: Sending request with id 3 (DEVICE_CONFIG)
[15:08:40][D][opentherm:372]: 00000000 00000011 00000000 00000000
[15:08:40]type: READ_DATA; id: 3; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[15:08:40][W][opentherm:406]: Protocol error occured while receiving response: type: NO_CHANGE_TOO_LONG; data: 00000000; clock: 1; capture: 00000000000000000000000111111111; bit_pos: 0
[15:08:40][D][opentherm:370]: Sending request with id 3 (DEVICE_CONFIG)
[15:08:40][D][opentherm:372]: 00000000 00000011 00000000 00000000
[15:08:40]type: READ_DATA; id: 3; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[15:08:40][W][opentherm:406]: Protocol error occured while receiving response: type: NO_CHANGE_TOO_LONG; data: 00000000; clock: 1; capture: 00000000000000000000000111111111; bit_pos: 0
[15:08:40][D][opentherm:370]: Sending request with id 3 (DEVICE_CONFIG)
[15:08:40][D][opentherm:372]: 00000000 00000011 00000000 00000000
[15:08:40]type: READ_DATA; id: 3; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[15:08:40][W][opentherm:406]: Protocol error occured while receiving response: type: NO_CHANGE_TOO_LONG; data: 00000000; clock: 1; capture: 00000000000000000000000111111111; bit_pos: 0
[15:08:41][D][opentherm:370]: Sending request with id 3 (DEVICE_CONFIG)
[15:08:41][D][opentherm:372]: 00000000 00000011 00000000 00000000
[15:08:41]type: READ_DATA; id: 3; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[15:08:41][W][opentherm:406]: Protocol error occured while receiving response: type: NO_CHANGE_TOO_LONG; data: 00000000; clock: 1; capture: 00000000000000000000000111111111; bit_pos: 0
[15:08:41][D][opentherm:370]: Sending request with id 3 (DEVICE_CONFIG)
[15:08:41][D][opentherm:372]: 00000000 00000011 00000000 00000000
[15:08:41]type: READ_DATA; id: 3; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[15:08:41][W][opentherm:406]: Protocol error occured while receiving response: type: NO_CHANGE_TOO_LONG; data: 00000000; clock: 1; capture: 00000000000000000000000111111111; bit_pos: 0
[15:08:41][D][opentherm:370]: Sending request with id 3 (DEVICE_CONFIG)
[15:08:41][D][opentherm:372]: 00000000 00000011 00000000 00000000
[15:08:41]type: READ_DATA; id: 3; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[15:08:33][W][opentherm:406]: Protocol error occured while receiving response: type: NO_CHANGE_TOO_LONG; data: 00000000; clock: 1; capture: 00000000000000000000000111111111; bit_pos: 0
[15:08:33]
[15:08:33]User exception (panic/abort/assert)
[15:08:33]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[15:08:33]
[15:08:33]Unhandled C++ exception: OOM
[15:08:33]
[15:08:33]>>>stack>>>
[15:08:33]
[15:08:33]ctx: sys
[15:08:33]sp: 3fffee30 end: 3fffffb0 offset: 0010
[15:08:33]3fffee40:  3fffbc9c 4021e9fb 00000200 4023aeb4  
[15:08:34]3fffee50:  00000030 3fffa4c0 3fffb77c 4021e945  
[15:08:34]3fffee60:  40100000 00000001 3fffb42c 4021ea0b  
[15:08:34]3fffee70:  00000014 00000000 00000001 ffffffff  
[15:08:34]3fffee80:  40000f3d 3fffbc9c 00000002 3fff3900  
[15:08:34]3fffee90:  3fffb7c4 3fff3900 3fffb7c4 3fff3ba8  
[15:08:34]3fffeea0:  3fffbc9c 3fff3bac 3fff3bad 40274a48  
[15:08:34]3fffeeb0:  00000000 3fffdad0 3fff34f8 00000030  
[15:08:34]3fffeec0:  00000010 00000000 00000000 3fff9d84  
[15:08:34]3fffeed0:  3fff0000 0000003c 00000020 3fff3900  
[15:08:34]3fffeee0:  3fffb7c4 3fffb7e2 3fff3b64 40278a99  
[15:08:34]3fffeef0:  00000014 00000000 3fff3900 40101082  
[15:08:34]3fffef00:  40294adc 3fffa4b4 3fff9d0c 3fff9d84  
[15:08:34]3fffef10:  3fff3900 00000008 3fffb7c4 40271935  
[15:08:34]3fffef20:  3fffdc80 3fff9d0c 3fffa4b4 40271758  
[15:08:34]3fffef30:  40297c91 3fff9d0c 3fffa4b4 40297ca3  
[15:08:34]3fffef40:  3fffb7d4 3fffb7c4 00000002 3fffdab0  
[15:08:34]3fffef50:  4029470b 00000000 3fffa4b4 40298f7b  
[15:08:34]3fffef60:  40000f49 3fffdab0 3fffdab0 40000f49  
[15:08:34]3fffef70:  40000e19 000b9035 bff00000 0000bfff  
[15:08:34]3fffef80:  00000005 aa55aa55 000000ed 40105bd5  
[15:08:34]3fffef90:  40105bdb bff00000 0000bfff ffffffff  
[... 525 lines cut ...]
  - platform: pid
    name: "Central heating"
    id: central_heating
    heat_output: t_set
    default_target_temperature: 19.5°C
    sensor: ch_room_temperature
    control_parameters: 
      kp: 0.81024 #0.23150 #0.38544
      ki: 0.0002 #0.00006 #0.00010
#      kd: 984.4467 #585.98022 #600.8665 #700.98022 #585.98022 #984.44672 #1040.70081 #585.98022
      min_integral: 0
      max_integral: 0.25
      starting_integral_term: 0
      output_averaging_samples: 1
      derivative_averaging_samples: 5
    deadband_parameters:
      threshold_high: 0.2
      threshold_low: -0.1
      kp_multiplier: 0.0
      ki_multiplier: 0.5
      kd_multiplier: 0.0
    visual:
      min_temperature: 10.0
      max_temperature: 25.0
`

olegtarasov, 2024-09-02:
Hi @Alfagek! It would be really helpful if you wrapped your yaml config in a code block :)

As I can gather from your logs, you get exactly the same faulted response on every communication with the boiler. The reason for this is unclear, and I would suggest the following:

1. Make sure you configured your IN and OUT pins correctly. It's a common pitfall, but the pin marked as `IN` on the board, must be configured as `out_pin` in YAML config, and vice versa. It seems like you are using pin numbers that are different from DIYLESS sample sketch, so I would suggest to double check.
2. This component is not extensively tested on `ESP8266`. Actually, only one person contacted me before about running this component on `ESP8266`, and I had to make some fixes in order to make it work. It worked for him, but there is certainly not enough data to draw conclusions. Anyway, I would really suggest using `ESP32` :)
3. If using `ESP32` is not an option, I would also try to reduce the number of components in your config. `ESP8266` has very limited resources, and there are some out of memory errors in your log. I would get rid of `web_server` and `captive_portal` components to save memory.

Alfagek, 2024-09-02:
Hi @olegtarasov ,

Thanks for the reaction back.

Stupid me about the pins, it was because of trying all the things to get it work again.
Its now back to pin 4/5.

It's using a lot more memory now. I did out comment the "web server" and "captive portal" and its working now. Sometime the buffer is running over, and crashes the D1. So when i have some time, i will make a esp32 mini ready to put it on. But before i didn't have this behavior, but it looks also that the data is getting readed quicker then it was (if i show the logs).

About the Yaml i posted, i did put it in a code block, but for a strange reason it would not take it, thats why it did so much editing ;-)

Thanks again for the feedback and keep up the good work (y)

Alfagek, 2024-09-07:
Hallo @olegtarasov,

Did changed to a esp32mini. Only thing that i can't get right is the Pump/Burner stats (id 116 till 122)

I think i do see the read/decode error's in my logs (Marked as **Bold**. But you are more a "specialist" with the opentherm library.

The errors are "Opentherm 406 and 410". With my old setup, i did get the stats, only they give sometimes a error. But now its keep staying on 0

**See logs below**

`[16:50:29][W][opentherm:410]: Receive response timed out at a protocol level

[16:50:29][D][opentherm:371]: Sending request with id 28 (RETURN_WATER_TEMP)
[16:50:29][D][opentherm:372]: 00000000 00011100 00000000 00000000
type: READ_DATA; id: 28; HB: 0; LB: 0; uint_16: 0; float: 0.000000
**[16:50:29][W][opentherm:406]: Protocol error occured while receiving response: type: NO_TRANSITION; data: 00x70071; clock: 1; capture: 00000000000000000000000000111111; bit_pos: 19**
[16:50:29][D][opentherm:371]: Sending request with id 28 (RETURN_WATER_TEMP)
[16:50:29][D][opentherm:372]: 00000000 00011100 00000000 00000000
type: READ_DATA; id: 28; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[16:50:29][D][opentherm:189]: Received OpenTherm response with id 28 (RETURN_WATER_TEMP)
[16:50:29][D][opentherm:190]: 00000100 00011100 00100110 00000000
type: READ_ACK; id: 28; HB: 38; LB: 0; uint_16: 9728; float: 38.000000
[16:50:29][D][opentherm:371]: Sending request with id 119 (DHW_BURNER_STARTS)
[16:50:30][D][opentherm:372]: 00000000 01110111 00000000 00000000
type: READ_DATA; id: 119; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[16:50:30][D][opentherm:189]: Received OpenTherm response with id 119 (DHW_BURNER_STARTS)
[16:50:30][D][opentherm:190]: 00000111 01110111 00000000 00000000
**type: UNKNOWN_DATAID; id: 119; HB: 0; LB: 0; uint_16: 0; float: 0.000000**
[16:50:30][D][sensor:094]: 'Boiler Number of starts burner during DHW mode': Sending state 0.00000  with 0 decimals of accuracy
[16:50:30][D][opentherm:371]: Sending request with id 17 (MODULATION_LEVEL)
[16:50:30][D][opentherm:372]: 00000000 00010001 00000000 00000000
type: READ_DATA; id: 17; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[16:50:30][D][opentherm:189]: Received OpenTherm response with id 17 (MODULATION_LEVEL)
[16:50:30][D][opentherm:190]: 00000100 00010001 00000000 00000000
type: READ_ACK; id: 17; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[16:50:30][D][sensor:094]: 'Boiler Relative modulation level': Sending state 0.00000 % with 2 decimals of accuracy
[16:50:30][D][opentherm:371]: Sending request with id 57 (MAX_CH_SETPOINT)
[16:50:30][D][opentherm:372]: 00000001 00111001 00110111 00000000
type: WRITE_DATA; id: 57; HB: 55; LB: 0; uint_16: 14080; float: 55.000000
[16:50:30][D][opentherm:189]: Received OpenTherm response with id 57 (MAX_CH_SETPOINT)
[16:50:30][D][opentherm:190]: 00000101 00111001 00110111 00000000
type: WRITE_ACK; id: 57; HB: 55; LB: 0; uint_16: 14080; float: 55.000000
[16:50:30][D][sensor:094]: 'Boiler Maximum allowable CH water setpoint': Sending state 55.00000 °C with 2 decimals of accuracy
[16:50:30][D][opentherm:371]: Sending request with id 56 (DHW_SETPOINT)
[16:50:30][D][opentherm:372]: 00000001 00111000 00110111 00000000
type: WRITE_DATA; id: 56; HB: 55; LB: 0; uint_16: 14080; float: 55.000000
[16:50:30][D][sensor:094]: 'ch_room_temperature': Sending state 22.30000  with 1 decimals of accuracy
[16:50:30][D][opentherm.output:008]: Received state: 0.00. Min value: 5.00, max value: 100.00
[16:50:30][D][opentherm.output:011]: Output t_set_t_set set to 0.00
[16:50:30][D][opentherm:189]: Received OpenTherm response with id 56 (DHW_SETPOINT)
[16:50:30][D][opentherm:190]: 00000101 00111000 00110111 00000000
type: WRITE_ACK; id: 56; HB: 55; LB: 0; uint_16: 14080; float: 55.000000
[16:50:30][D][sensor:094]: 'Boiler Domestic hot water temperature setpoint': Sending state 55.00000 °C with 2 decimals of accuracy
[16:50:30][D][opentherm:371]: Sending request with id 19 (DHW_FLOW_RATE)
[16:50:30][D][opentherm:372]: 00000000 00010011 00000000 00000000
type: READ_DATA; id: 19; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[16:50:31][D][opentherm:189]: Received OpenTherm response with id 19 (DHW_FLOW_RATE)
[16:50:31][D][opentherm:190]: 00000100 00010011 00000000 00000000
type: READ_ACK; id: 19; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[16:50:31][D][sensor:094]: 'Boiler Water flow rate in DHW circuit': Sending state 0.00000 l/min with 2 decimals of accuracy
[16:50:31][D][opentherm:371]: Sending request with id 117 (CH_PUMP_STARTS)
[16:50:31][D][opentherm:372]: 00000000 01110101 00000000 00000000
type: READ_DATA; id: 117; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[16:50:31][D][opentherm:189]: Received OpenTherm response with id 117 (CH_PUMP_STARTS)
[16:50:31][D][opentherm:190]: 00000111 01110101 00000000 00000000
**type: UNKNOWN_DATAID; id: 117; HB: 0; LB: 0; uint_16: 0; float: 0.000000**
[16:50:31][D][sensor:094]: 'Boiler Number of starts CH pump': Sending state 0.00000  with 0 decimals of accuracy
[16:50:31][D][opentherm:371]: Sending request with id 25 (FEED_TEMP)
[16:50:31][D][opentherm:372]: 00000000 00011001 00000000 00000000
type: READ_DATA; id: 25; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[... 114 lines cut ...]
[16:50:40][D][opentherm:372]: 00000000 00010010 00000000 00000000
type: READ_DATA; id: 18; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[16:50:40][D][sensor:094]: 'ch_room_temperature': Sending state 22.30000  with 1 decimals of accuracy
[... cut ...]
[16:50:41][D][opentherm:372]: 00000000 00011100 00000000 00000000
type: READ_DATA; id: 28; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[16:50:42][W][opentherm:410]: Receive response timed out at a protocol level
[16:50:42][D][opentherm:371]: Sending request with id 28 (RETURN_WATER_TEMP)
[16:50:42][D][opentherm:372]: 00000000 00011100 00000000 00000000
type: READ_DATA; id: 28; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[16:50:42][D][opentherm:189]: Received OpenTherm response with id 28 (RETURN_WATER_TEMP)

`

Thanks in advanced again.

olegtarasov, 2024-09-10:
Hi @Alfagek!

I see several issues in your log:

1. A couple of `Protocol error` occurences. This is completely OK, intermittent protocol errors happen from time to time and pose no harm. Actually, the more different components you have in your YAML config, the more errors occur. My theory is that ESPHome components (especially Dallas temperature sensors) are very liberal with disabling interrupts and it messes with OpenTherm protocol handling. You can improve the situation by specifying `sync_mode: true` in your `opentherm:` section, but these error will never go away completely.
2. A couple of `UNKNOWN_DATAID` errors. Looks like your boiler simply doesn't support these types of messages. Boiler responses seem well-formed (`UNKNOWN_DATAID` and other fields are zeroes), so not much we can do here. I don't know why exactly did you get actual values with other libraries — maybe your boiler reports them from time to time, maybe there is a bug in that other library. Seeing that other sensors and outputs work OK in your setup, there is not much to go on here really 🤷

Alfagek, 2024-09-28:
Hello Oleg,

Sorry for the late reaction. Did have a lot off projects at home.

Thanks or the explanation. 

About the boiler errors, i think indeed it was a problem in the old libraries, because it now don't give me numbers. And there before it gave. But it was going up and down. And thats strange for example burner hours ;-)

Keep up the good work!

closed 2024-09-28 by Alfagek