olegtarasov/esphome-opentherm #8: Possible to use as gateway?

kalua85, 2024-09-19:
Is it possible or are there plans to make to be able to use this library in gateway mode? I want to monitor and sometimes maybe adjust the messages between my thermostat and boiler.

olegtarasov, 2024-09-19:
Hi @kalua85! Yes, this is exactly what I'm going to work on after this component is merged into ESPHome. I can't give you an estimate, but I want this feature myself, so I will try and make time for it :)

kalua85, 2024-09-20:
Ah perfect, I used https://github.com/Reproduktor/esphome-OpenthermGW in the past, but I found it highly unstable and many messages where missed due to timing issues I guess. But maybe it can give you some inspiration

Minims, 2024-11-25:
I would love this too. Currently testing the thermostat version (master) :-)