olegtarasov/esphome-opentherm #9: Switch to set bit 6 in master status (DHW block)

rmull561, 2024-09-19:
Great work on getting this into esphome! I have a small feature request if it's ok to ask here.

My boiler (De Dietrich) uses bit 6 instead of bit 1 in the master status byte to enable/disable DHW. The bit 1 only disables an eco mode (DHW pre-heating). Someone mentioned this bit as a new thing added in OT3.0 [here](https://github.com/ihormelnyk/opentherm_library/issues/1#issuecomment-965124824_).

Can you please consider adding a switch for this bit, maybe as "dhw_block"?

olegtarasov, 2024-09-19:
Hi @rmull561! Sure, no problem. I will add this switch in a couple of days along with other switches and sensors people requested over time.

rmull561, 2024-10-01:
Yes, the DHW block works. Thanks!

It took me some time to figure out I need to also add dhw_block: yes to the opentherm section. Could it be enabled automatically when referenced by a switch?

I'm not sure if it's related to switching to this new component or disabling the permanent "eco" mode in the boiler's menu, but I noticed that the boiler now briefly activates with every restart of esphome.

olegtarasov, 2024-10-01:
@rmull561 Can you collect full debug logs to help me diagnose both issues? The best way to do it is to connect to ESP32 with a cable and run the firmware with `esphome run` — that way you will get all the logs from the start of the MCU.
I would need the logs for both cases: when `dhw_block` is used only as a switch, and when it is specified both as a switch and at the hub level.

rmull561, 2024-10-04:
Here are the logs:
[ot-log-hub.txt](https://github.com/user-attachments/files/17256958/ot-log-hub.txt)
[ot-log-nohub.txt](https://github.com/user-attachments/files/17256959/ot-log-nohub.txt)

olegtarasov, 2024-10-25:
Hi @rmull561! Sorry it took some time to get back to this issue. Actually, the logic when you need to specify `true` at the hub level in order to be able to use a switch, is the original logic from Arthur Rumps' library. I'm not entirely sure whether it's a good idea to change how it works and introduce a breaking change to people's configurations. For now, I think it's better to leave it as it is.

As for boiler activating on ESPHome restart, can you test the latest commit I've pushed to `develop`? It ensures that `STATUS` message is sent to the boiler as early as possible.

In order to test the `develop` branch, you need to change your `external_components` declaration:

```yaml
external_components:
  source: github://olegtarasov/esphome-opentherm@develop
```

rmull561, 2024-10-26:
I ran a few tests with the develop branch. It didn't seem to help. But I remember it used to work before, so I did more experiments and I think I finally understand what is happening. It's the flashing process that can take so long that the boiler falls back to the relay on/off mode and the preheating activates. More complex esphome code compiles into bigger binary and takes more time to upload. I suspect when new binary is being uploaded, esphome stops communicating with the boiler and there is nothing we can do about that.

As for the required settings to true in the hub, could all of those be shown in the documentation?

Anyway, I'm happy with the new switch setting the bit 6. I think this issue can be closed. Thanks.

olegtarasov, 2024-10-29:
I didn't realize that the problem occurred during flashing. This is expected for exactly the reasons you described :)

closed 2024-10-29 by olegtarasov