olegtarasov/esphome-opentherm #10: Fetch t_dhw_set and max_t_set when configured as numeric input

takerukoushirou, 2024-10-12:
Thank you for the hard work on refining this component into an official ESPHome component!

I was experimenting with what sensors are supported by my heater (which sadly reports zero for many measurements that it certainly internally does) and what values can be set. Setting DHW and maximum CH setpoints is well supported and reflected on the device. I noticed an inconsistency though. When the two number inputs are defined, they show up in HA and can be set, but they initially start with blank inputs. If I instead configure sensors for `t_dhw_set` and `max_t_set`, they show the currently on the heater configured values. Defining both the number inputs and sensors again leads to initially blank inputs but also 0.0 as reported values for both sensors. It seems that when number inputs are configured, reading `t_dhw_set` and `max_t_set` is disabled or fails.
It would be ideal if the number inputs could initially show the values configured on the heater.

Corresponding configuration segments:
```yaml
output:
  - platform: opentherm
    t_set:
      id: opentherm_ch_set_temperature
      min_value: 30
      max_value: 80
      auto_max_value: true
      zero_means_zero: true

number:
  - platform: opentherm
    t_dhw_set:
      name: "Domestic hot water setpoint"
      auto_min_value: true
      auto_max_value: true
    max_t_set:
      name: "Central heating maximum setpoint"
      auto_min_value: true
      auto_max_value: true

# Sensors work when no number input is specified, otherwise read 0.0.
sensor:
  - platform: opentherm
    t_dhw_set:
      name: "Domestic hot water setpoint"
    max_t_set:
      name: "Central heating maximum setpoint"
```

[image]

olegtarasov, 2024-11-28:
Hi @takerukoushirou @dwulkiewicz! Sorry, it took some time to look into this issue. Unfortunately, the behavior of not being able to read a value while output is configured, is baked into component's architecture and can't be changed easily. I will keep this issue in mind, but I can't make any promises that it will become a priority.

dwulkiewicz, 2024-11-28:
Thank you for your answer.

The assumptions are as follows: max_t_set and t_dhw_set are controlled from HA and the boiler panel. When first started, the ESPHome controller reads these settings from the boiler. Then they can be changed either from HA or from the boiler panel. To do this, the action of setting these values ​​can only occur after calling id(max_t_set_output).set_level() and the reading of the max_t_set and t_dhw_set values ​​from the boiler must be available.

Now it is not possible to change these parameters from the boiler panel because they are immediately overwritten with values ​​from HA.

This is even more than @takerukoushirou described, who only talks about reading the values ​​on the first start.

I hope I have described the problem well and it can be changed eventually.

Best regards