olegtarasov/esphome-opentherm #11: Daikin D2C

dmd79, 2024-10-19:
Hello all...
I'm trying to use esphome opentherm on my Daikin D2C boiler
I can se many parameter and randomly turn on and off boiler, but seems something is missing
With original Daikin thermostat I can see on boiler display the room temperature, but when using esp32 the boiler display show ---
Seems the original thermostat send to the boiler the room temperature and I think the boiler has integrated PID
If you have some idea about this, any test I can do please tell me
Thank

dmd79, 2024-10-22:
Today I have properly controlled t_set and boiler temperature was right.. set to 45°, then to 60° and modulation rise up... Set to 30° and modulation fall down near to 0%. Then turn off boiler central heating and turn on again... Boiler never turn on again... My OT version is 4 according to opentherm_version_device sensor

olegtarasov, 2024-10-22:
Hi @dmd79! Unfortunately, every boiler is different and all I can do is just look at the logs and see if any errors are reported. Obviously, you've already done that. I have a really basic Buderus boiler which supports the bare minimum of OpenTherm protocol, so I can't even comment on how most of the sensors and controls should normally be working.

As for your boiler not turning central heating back on — did you try to power down the boiler, disconnect the OpenTherm bridge and run the boiler without the thermostat? On my boiler I need to short OpenTherm terminals in order to signal that thermostat is not present.

dmd79, 2024-10-22:
Hi and thank for the answer...
I have understand the pid issue.. I was using an external home assistant temperature sensor and even show correct value on esp web page it was for some reason bad for pid. Now I'm using and adc input with a  potentiometer and pid work as it should.. The only issues remain the DHW control but it's a minor issue

FreeBear-nc, 2024-10-25:
With some boilers, the controller (master) needs to announce the OT version that it implements using ID124 if I remember my notes correctly. Failure to announce OT version could be the reason @dmd79 is having problems.

olegtarasov, 2024-10-25:
@FreeBear-nc, thank you! Will implement this message in the `develop` branch and test it out.

olegtarasov, 2024-10-25:
Note to self: found an example usage here: https://github.com/Laxilef/OTGateway/blob/351a884685d92884c55b0e0a591911ba3ed5557b/src/OpenThermTask.h#L810

dmd79, 2024-10-25:
> With some boilers, the controller (master) needs to announce the OT version that it implements using ID124 if I remember my notes correctly. Failure to announce OT version could be the reason @dmd79 is having problems.

Thank for your help. Looking here seems the right I'd
https://github.com/ihormelnyk/opentherm_library/blob/master/src/OpenTherm.h#L144

olegtarasov, 2024-11-04:
@dmd79 I've added an ability to specify master OpenTherm version in `develop` branch. You can reference it as follows:

```yaml
external_components:
  source: github://olegtarasov/esphome-opentherm@develop
  refresh: 0s
```

You can specify OpenTherm version as a float at hub level:

```yaml
opentherm:
  in_pin: IN_PIN
  out_pin: OUT_PIN
  opentherm_version: 2.2   # I don't know what version do you need, my boiler doesn't support this message :)
```

dmd79, 2024-11-19:
Hello, I've got the log with otgw
[d2c.log](https://github.com/user-attachments/files/17822736/d2c.log)
in this log I have power on boiler and then set dhw on Daikin thermostat to 52 50 48 46 44 42 49 38 36 35 degree
If I'm not wrong my thermostat is writing dhw setpoint to ID 162

-> WriteData 162 34 0
-> WriteData 162 32 0
-> WriteData 162 30 0
-> WriteData 162 2E 0
-> WriteData 162 2C 0
-> WriteData 162 2A 0
-> WriteData 162 28 0
-> WriteData 162 26 0
-> WriteData 162 24 0
-> WriteData 162 23 0

the f8.8 value of 52 50 48 46 .... should be 34 32 30 2E.....

dmd79, 2024-11-20:
Ok I'm right. The ID is 162 and the easy fix is this, but it can't be done for other of course
https://github.com/dmd79/esphome-opentherm/commit/ce7724d3f044d702738834e4864f2d77072d601d
would be fine if you can the possibility to change the ID on esphome frontend

olegtarasov, 2024-11-20:
Hi @dmd79! I will look into adding an ability to override message id in a day or two.

olegtarasov, 2025-01-06:
I added an ability to override message IDs, so closing this issue.

closed 2025-01-06 by olegtarasov