olegtarasov/esphome-opentherm #12: Curious values from Fan Speed sensor

droscy, 2024-10-24:
Hello olegtarasov, first of all I would thank you for the great work you are doing in having this component merged in esphome, really appreciated!

I have curious values from Fan Speed sensor, il reports 16896 rpm when the boiler is idle and it decreses when the boiler is running.

[image]

These are the log messages when the boiler is idle

    [06:51:52][D][opentherm:299]: Sending request with id 35 (FAN_SPEED)
    [06:51:52][D][opentherm:300]: 00000000 00100011 00000000 00000000
    type: READ_DATA; id: 35; HB: 0; LB: 0; uint_16: 0; float: 0.000000
    [06:51:52][D][opentherm:129]: Received OpenTherm response with id 35 (FAN_SPEED)
    [06:51:52][D][opentherm:130]: 00000100 00100011 01000010 00000000
    type: READ_ACK; id: 35; HB: 66; LB: 0; uint_16: 16896; float: 66.000000
    [06:51:52][D][sensor:094]: 'Fan Speed': Sending state 16896.00000 RPM with 0 decimals of accuracy

I will add log messages during boiler run later (I'm collecting them right now).

I don't know if I can simply filter the values computing `16896 - ${value}`.

I will read throught the boiler manual searching for a way to have the fan speed displayed by the boiler monitor itself, so I can compare those values to values reported by esphome. I'll keep you updated.

olegtarasov, 2024-10-24:
Hi @droscy! I couldn't find proper documentation for message id 35, so it's possible it was implemented incorrectly in the original library. Some xls document that is circulating on the internet says id 35 is „Boiler fan speed setpoint and actual value”, so that already suggests that this value shouldn't be interpreted as a single unsigned int like it does now. Can you post more logs for message id 35 when the boiler is active? I suspect that actual RPM value is reported only in the high byte.

droscy, 2024-10-24:
These are the logs during DHW active

    [07:17:47][D][opentherm:299]: Sending request with id 35 (FAN_SPEED)
    [07:17:47][D][opentherm:300]: 00000000 00100011 00000000 00000000
    type: READ_DATA; id: 35; HB: 0; LB: 0; uint_16: 0; float: 0.000000
    [07:17:47][D][opentherm:129]: Received OpenTherm response with id 35 (FAN_SPEED)
    [07:17:47][D][opentherm:130]: 00000100 00100011 00011101 00011111
    type: READ_ACK; id: 35; HB: 29; LB: 31; uint_16: 7455; float: 29.121094
    [07:17:47][D][sensor:094]: 'Fan Speed': Sending state 7455.00000 RPM with 0 decimals of accuracy

The display in the boiler was reporting 1800 rpm approximately for the same time.

Here https://pastebin.com/GUxyn7Tc you can find the whole FAN_SPEED log, since DHW activation at 07:16:17 to DHW switch off at 07:19:06.

I'll try to grab more comparisons between logs and display.

droscy, 2024-10-24:
The xls you found can be valid: my logs may suggest that HB is the target value and LB is the actual value, during boiler active, because every time HB changes, LB tries to "reach" HB.

FreeBear-nc, 2024-10-24:
I have interpreted ID35 as being a u16 value, and when divided by 60, gives the true speed in RPM.
Have asked the OpenTherm Association for a copy of the latest specification (it is shown as a download on their web site), but am waiting for a response.

olegtarasov, 2024-10-25:
@FreeBear-nc, thanks! Is 60 a value that you inferred, or is it documented somewhere? It doesn't seem to be valid for @droscy since his reported value is decreasing when the fan spins faster.

FreeBear-nc, 2024-10-25:
I had come across a description of ID84 & ID85 where it claimed true RPM=value/60 and made the assumption that ID35 would be the same.
When I first started using hor Melnyk's library (with extensions) with my boiler, I was getting some pretty wild numbers when reading ID35 and assumed a divisor of 60. However, popping the cover off my boiler and looking at the label on the fan, it states maximum speed is 14200RPM :open_mouth:
Looking at the specs for the [ebmpapst](https://www.ebmpapst.com/gb/en/products/blowers/gas-blowers.html) gas blowers, they all seem to have a top speed of 6000 to 14000 RPM.

It could be that the response @droscy is getting should be interpreted as ID35:LB * 60 - If we knew the make & model of the boiler and the fan, it might be possible to make an educated guess as to how to interpret the data. If it turns out some boilers are returning a u16 value and others, a u8, it is going to get confusing.

droscy, 2024-10-25:
My boiler is **Alhena Tech 24 C** (0T4B2AWD) by _Lamborghini Caloreclima_. I don't know the maker/model of the fan, I'll ask my installer.

FreeBear-nc, 2024-10-27:
> Hi @droscy! I couldn't find proper documentation for message id 35, so it's possible it was implemented incorrectly in the original library. Some xls document that is circulating on the internet [...]

Had hoped that I could measure the fan speed directly on my boiler, but no access to any moving parts.
Looking at various sources including Ihor Melnyk's opentherm library, they all say "fan speed set point and actual value" implying actual value is the LB. As this only gives a range of 0 to 255, multiplying by 60 to get RPM seems to be reasonable.

Splitting ID35 in to HB and LB values, the two numbers track very closely with my boiler and the LB lags slightly behind HB. That matches what @droscy has observed, and if we assume the values are Revs Per Second, it fits with some of my assumptions.

olegtarasov, 2024-10-29:
Hi @droscy! I've merged a pull request from @FreeBear-nc that changes how FAN_SPEED message is processed. You can try recompiling your configuration from the latest commit in `main` and adding a `multiply` by 60 filter to fan speed sensor.

FreeBear-nc, 2024-10-29:
Just pushed a small edit to hub.cpp and schema.py to automatically multiply the return value. This saves having to use a filter in the sensor.yaml and is based on the assumption that IB35 hb/lb values are revs per second.

droscy, 2024-10-30:
Hello @olegtarasov, the measured values are now correct or at least they almost match what is reported by the boiler's display. Many thanks.

As suggested by @FreeBear-nc I think that the multiplication should be executed inside the library and not by the user in the configuration.

 Just one side note: I don't remember the icon of the fan speed in Home Assistant before the merge, but now both `fan_speed` and `fan_speed_setpoint` are using the default `eye` icon, I would use `mdi:fan`.

olegtarasov, 2024-10-31:
@droscy I've merged another PR from @FreeBear-nc which should fix the icon and eliminate the need for a multiply filter. Can you verify that it works?

FreeBear-nc, 2024-10-31:
Was thinking about this last night... Two options:
fans_speed remains u8 with the multiplier, and a new fan_speed_raw (or fan_speed_u16) for the u16 cases.
Or fan_speed reverts to u16, and we have a fan_speed_lb (with, or without the multiplier).

The alternative is to have a yaml option to select either u8 or u16 values and let the python script automatically generate the relevant code.

droscy, 2024-10-31:
I would suggest having a config option, of course I'll let you choose, but I think it is better to have a single `fan_speed` sensor and an option to select the "mode" that value is reported.

The new config could be at `opentherm` level or at `fan_speed` level: the former allows the code to raise a warning in case the user choose `u16` fan speed mode and have the `fan_speed_setpoint` sensor in the yaml too.

FreeBear-nc, 2024-10-31:
```
sensor:
  - platform: opentherm
    fan_speed:
      name: "Boiler fan speed"
      message_data: "u16" # overrides the default u8_lb_60 message

    fan_speed_setpoint:
      name: "Boiler fan speed set point"
      message_data: "u8_hb_60"
```

Not that familiar with the python code is esphome, so would struggle to implement this.
Having a fan_speed_u16 sensor would be quick & simple.

FreeBear-nc, 2024-11-03:
> @FreeBear-nc I've been contacted by another user of this integration, and the last change broke his fan speed sensor. Turns out, sometimes fan speed is indeed should be interpreted as `u16` with no multiplication. Looks like it's best to revert `fan_speed` to its original `u16` handler and add an alternative sensor that uses `u8`.

Raised a PR that adds message_data to the sensor yaml. Took a while to get there, but I think it works...

FreeBear-nc, 2024-11-04:
> I think that having an option to override the data type is better since we might encounter other sensors that don't follow the spec in the future.
My thoughts exactly. It could be used to extract a set of flags and then processed in a lambda... Perhaps eventually have a yaml line that sets the MESSAGE as a raw number and allow for as yet undefined IDs to be processed.

olegtarasov, 2024-11-05:
The issue seems resolved, closing.

closed 2024-11-05 by olegtarasov

droscy, 2024-11-06:
Sorry guys, I didn't find time to test last commit, I'll test it as soon as possibile. However thanks again for your work.
I would remind you to update README/documentation for the new `message_data` option along with all its available values. Thanks.

olegtarasov, 2024-11-06:
@droscy Thanks! I've already added a section on `fan_speed`: https://github.com/olegtarasov/esphome-opentherm#fan-speed-sensor. I'm hesitant to announce the fact that every sensor has `data_type` property now, so that people don't abuse it.