olegtarasov/esphome-opentherm #14: HVAC devices

aluckos, 2024-10-31:
Hi Oleg. It's ratcher question not issue, are You planning to cover code for Hvac devices ? 
I found this part at headers, propably shoud be usefull to control ventilation systems.
Im trying to run my ventlation with some opentherm shield board but my programming skills are to poor to achive goal.

// HVAC Specific Message IDs
  HVAC_STATUS = 70,
  REL_VENT_SETPOINT = 71,
  DEVICE_VENT = 74,
  HVAC_VER_ID = 75,
  REL_VENTILATION = 77,
  REL_HUMID_EXHAUST = 78,
  EXHAUST_CO2 = 79,
  SUPPLY_INLET_TEMP = 80,
  SUPPLY_OUTLET_TEMP = 81,
  EXHAUST_INLET_TEMP = 82,
  EXHAUST_OUTLET_TEMP = 83,
  EXHAUST_FAN_SPEED = 84,
  SUPPLY_FAN_SPEED = 85,
  REMOTE_VENTILATION_PARAM = 86,
  NOM_REL_VENTILATION = 87,
  HVAC_NUM_TSP = 88,
  HVAC_IDX_TSP = 89,
  HVAC_FHB_SIZE = 90,
  HVAC_FHB_IDX = 91,

Best regards Artur

olegtarasov, 2024-11-04:
Hi @aluckos! Unfortunately, I have no experience in working with OpenTherm HVAC systems, and I don't have access to such a system to test things out. In theory, this component implements a generic OpenTherm protocol, which is not limited to boilers. So if HVAC system in question supports OpenTherm, it may just work out of the box.

olegtarasov, 2024-11-28:
Closing this issue since there is not much action I can take here.

closed 2024-11-28 by olegtarasov