olegtarasov/esphome-opentherm #15: Random disconnects

FreeBear-nc, 2024-11-13:
Just a heads up.
Have been running the opentherm component now that it has been fully merged in to esphome. Left it running this evening and have been experiencing random disconnects. Have failed to record any meaningful messages, but did see something flash past suggesting at least one event was triggered by an exception - This _might_ be a result of using `sync_mode: true` - Certainly getting a lot of warnings about `Component opentherm took a long time for an operation (216 ms)`. Perhaps the watchdog is biting.
Unfortunately, my time is limited during the week, so won't have time to dive in deeper until the weekend. Hope to capture a crash dump using https://github.com/dentra/esphome-components/tree/master/components/crash_info

tbrasser, 2024-11-14:
Similar issue here, as soon as I add documented sensors and binary sensors it just goes offline afyer a while.

olegtarasov, 2024-11-14:
I'm experiencing random disconnects myself, but I didn't have time to get to the bottom of the issue. I decided that it could be an issue with cheap dev boards I'm using, or some other fluke. I'm using this component for over a year now and these disconnects seem to be harmless and don't affect my boiler. If anyone gets to the root cause of why this is happening, I would be happy to fix the problem if it indeed comes from this component.

FreeBear-nc, 2024-11-14:
Looking at my HA logs, the disconnects appear randomly between 1min39sec and 50min periods. Whilst the boiler is running, these disconnects cause a shutdown of 1 to 4 minutes duration. On starting up, the boiler runs at full (rate limited) power for ~30 seconds - This is wasteful of gas and puts undue stress on the mechanical components. So to my mind, undesirable and impacts on efficiency & long term reliability.
Have extended the OT cable so that I can run the ESP32 on the bench with a JTAG debugger attached - Will attempt to do some work on it over the weekend.

olegtarasov, 2024-11-15:
@FreeBear-nc It seems your boiler reacts differently than mine :( I've just initiated a forced restart while boiler was running, and flame did switch off for 1-2 seconds, but switched back on almost immediately. I think that's why I don't notice these effects. I've also checked HA logs and saw just one disconnect in a course of several days. So it would be great if could catch this problem on your end, since it doesn't seem to appear all that often for me.

As a side note, do you experience the same disconnects with your version of the component based on original Arthur Rump's repo?

tbrasser, 2024-11-15:
Are you running with a lot of sensors enabled? Seems it's more stable with fewer things enabled like in the example.

olegtarasov, 2024-11-15:
I have 8-9 sensors in my config, since my boiler doesn't support much of the OpenTherm spec. Normally, the number of sensors shouldn't affect stability as each esphome loop iteration sends only one message at a time.
Maybe your disconnects happen because you are using ESP8266 and not ESP32. This component wasn't thoroughly tested on ESP8266 and I generally recommend using ESP32.

tbrasser, 2024-11-15:
Let me swap to an esp32 and report back (will try to not hijack threads, but also didn't want to create duplicates, let me know what you prefer)

FreeBear-nc, 2024-11-15:
> As a side note, do you experience the same disconnects with your version of the component based on original Arthur Rump's repo?

No disconnects using Arthur's code - Using my own fork with a current version of Ihor Melnyk's OT library on a Wemos D1 mini (ESP8266 based). Have a selection of other boards I _could_ try, but I'd rather get to the bottom of this problem as I suspect many users will also be using the D1 mini.

FreeBear-nc, 2024-11-15:
> Let me swap to an esp32 and report back (will try to not hijack threads, but also didn't want to create duplicates, let me know what you prefer)

Lets keep to one thread rather than spreading the same issue across multiple reports.
Whilst I can drop an ESP32 board in (have S2 and C3 variants), it doesn't help D1-mini users.

FreeBear-nc, 2024-11-16:
OK. Identified the exception - `Unhandled C++ exception: OOM`
The culprit is stringstream in debug_data() and possibly debug_error(). Setting debugging level higher than DEBUG doesn't fix the problem. Have reworked the print statements, and will generate a PR shortly.

Still not fixed the problem - Am now getting an illegal instruction in start_read_timer().

FreeBear-nc, 2024-11-17:
> Still not fixed the problem - Am now getting an illegal instruction in start_read_timer().

Removing the crash_info* component seems to have fixed the illegal instruction (subject to further monitoring).

*) https://github.com/dentra/esphome-components/tree/master/components/crash_info

FreeBear-nc, 2024-11-18:
> Still not fixed the problem - Am now getting an illegal instruction in start_read_timer(). After removing the crash_info component, the illegal instruction has moved to verify_stop_bit().

Looks like the illegal instruction is caused by a faulty flash chip - Currently running on an ESP32-C3 board (with PR18 mods), and no errors yet. Have ordered some more D1-mini boards (both ESP32 and ESP8266 variants) and an ESP32-S3 for further testing. It will be about a week before they arrive.

olegtarasov, 2024-11-18:
@FreeBear-nc Thank you for doing such an awesome debugging job! I've also received a ESP8266 board and plan to run my own boiler with it starting from this weekend.

FreeBear-nc, 2024-11-24:
Been running with an ESP32-C3 over the last week without any crashes. New ESP8266 boards turned up yesterday. Still getting crashes from illegal instructions. Have been digging for answers, and am currently running with a provisional fix.

olegtarasov, 2024-11-25:
@FreeBear-nc Thank you for such a great work debugging this problem! I will propose a new PR to esphome with the most critical changes later today.

FreeBear-nc, 2024-11-26:
Reviewed HA history over the last week when I had been using an ESP32-C3. Disconnects were still happening after 10-70 hours of running. No logs saved to verify if it was an illegal instruction, although I suspect it was.
Raised a PR with esphome as it appears the relevant patch got dropped from PR7835.

olegtarasov, 2024-11-26:
I have a similar situation with vanilla ESP32: disconnects don't happen frequently, but they still appear. Will flash a fix with `IRAM_ATTR` and see what happen. `verify_stop_bit_` and `check_parity_` are used on ESP32 as well, so there is a real chance those crashes will be solved.

olegtarasov, 2024-11-28:
Both fixes got merged to esphome. I also pushed `v1.0.4` to this repo, which contains the fix. Closing this issue for now. @tbrasser feel free to reopen if you are still experiencing disconnects after applying the fix.

closed 2024-11-28 by olegtarasov