olegtarasov/esphome-opentherm #16: CH circuit also heated during DHW

tbrasser, 2024-11-14:
I had this before on the predecessors of this component (using latest esphome beta now)

on the non-component custom lambda version I do not experience this issue.

when I turn on the tap/shower:

- water gets warm
- then gets lukewarm while also radiators heat up

olegtarasov, 2024-11-14:
Hi @tbrasser! Can you post your full yaml config and debug logs for both components? Also, please provide a link to the custom lambda component you are using. I'll try and figure out what's the difference in behavior.

tbrasser, 2024-11-14:
I've reverted to my old config for now:

This is what I'm running now and works fine:
https://github.com/tbrasser/config/blob/main/esphome/includes/opentherm.yaml

It uses: https://github.com/rsciriano/ESPHome-OpenTherm/tree/main/esphome-opentherm

The issue of lukewarm DHW with CH heating up has occurred with all variants of the component opentherm, for example: https://github.com/tbrasser/config/blob/main/esphome/includes/opentherm_old.yaml

olegtarasov, 2024-11-15:
I wasn't able to analyze your config in detail yet, but now I see that you configured `t_dhw_set` as an output and there is no place where this output is being set. Generally, DHW setpoint should be configured as simple number, and `output` platform is mostly used with climate controllers (like PID regulators) to control central heating. So I'm almost sure that DHW will work for you with ESP8266 as well, when you configure DHW setpoint as number.

closed 2024-11-16 by tbrasser