olegtarasov/esphome-opentherm #19: Protocol error occured while receiving response: type: NO_CHANGE_TOO_LONG

nyctomanica, 2024-11-16:
Hey, thank you so much for developing this!

I've been trying to set up this component recently, using a Wemos D1 Mini esp32, thehognl opentherm shield, and a Viessmann 050-W combi boiler.

However I've been unable to get anything to work so far, with my logs filled with "Protocol error occured while receiving response: type: NO_CHANGE_TOO_LONG" every few ms.
My boiler also shows error A9, meaning Opentherm Error.

I've tried reversing the pins for in/out and simplifying my config file to a very basic level, but the issue persists.

Could you let me know if this is a bug, or user error/stupidity :)

My config is below, and I have attached my latest full logs.

Cheers!

```
esphome:
  name: heat-o-matic
  friendly_name: Heat-o-matic

esp32:
  board: wemos_d1_mini32

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: [redacted]

ota:
  - platform: esphome
    password: [redacted]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Heat-O-Matic Fallback Hotspot"
    password: [redacted]

captive_portal:

external_components:
  source: github://olegtarasov/esphome-opentherm@main

opentherm:
  in_pin: GPIO19
  out_pin: GPIO18

# Values that can be set maually from HA interface
number:
  - platform: opentherm
    # Hot water temperature
    t_dhw_set:
      id: dhw_setpoint
      name: "Hot water temperature"
      min_value: 30
      max_value: 60
      initial_value: 45
      restore_value: true
  # Heating loop temperature
    t_set:
      id: ch_setpoint
      name: "Heating loop temperature"
      min_value: 35
      max_value: 70
      initial_value: 45
      restore_value: true

switch:
  - platform: opentherm

    ch_enable:
      id: ch_enable
      name: "Central Heating enabled"
      restore_mode: RESTORE_DEFAULT_OFF
    dhw_enable:
      id: dhw_enable
      name: "Hot Water enabled"
      restore_mode: RESTORE_DEFAULT_OFF

sensor:
  - platform: opentherm
    rel_mod_level:
      name: "Relative modulation level"
    ch_pressure:
      name: "Water pressure"
    t_boiler:
      name: "Heating water temperature"
    t_dhw:
      name: "Hot water temperature"
    t_exhaust:
      name: "Exhaust temperature"
    t_ret:
      name: "Return water temperature"
    opentherm_version_device:
      name: "Opentherm version"
    oem_diagnostic_code:
      name: "Boiler diagnostic code"
    oem_fault_code:
      name: "Boiler fault code"

binary_sensor:
  - platform: opentherm
    ch_active:
      name: "Central Heating active"
    dhw_active:
      name: "Hot Water active"
    flame_on:
      name: "Burner on"
    fault_indication:
      name: "Boiler fault"
      entity_category: diagnostic
    diagnostic_indication:
      name: "Diagnostic event"
      entity_category: diagnostic
    service_request:
      name: "Service needed"
      entity_category: diagnostic
```

nyctomanica, 2024-11-16:
oops, forgot logs
[logs_heat-o-matic_logs.txt](https://github.com/user-attachments/files/17787979/logs_heat-o-matic_logs.txt)

FreeBear-nc, 2024-11-17:
> I've been trying to set up this component recently, using a Wemos D1 Mini esp32, thehognl opentherm shield, and a Viessmann 050-W combi boiler.
> ```
> opentherm:
>   in_pin: GPIO19
>   out_pin: GPIO18
> ```

Pretty sure GPIO18 & GPIO19 are not routed to pins on the D1-mini. Try:
```
opentherm:
  in_pin: GPIO12
  out_pin: GPIO14
```

Note: I've been experiencing random crashes using a D1-mini and have raised a PR that _should_ fix the problem. Also waiting for some more D1-mini boards to be delivered, as I might have a duff flash chip on mine..

FreeBear-nc, 2024-11-18:
Just ordered one of them D1-mini-ESP32 boards. Should take about a week to arrive from China. Once here, I can check the GPIO assignments.

clau-bucur, 2024-11-18:
I have the same D1 mini ESP32 board running fine with:
```
opentherm:
  in_pin: GPIO19
  out_pin: GPIO18
  ch_enable: true
  dhw_enable: true
```

I can check later the exact used pins on the board if needed.

nyctomanica, 2024-11-19:
Aye sure, here you go!
That's a good shout too, I'll try using a different style of ESP32 board (got some AZdelivery ones kicking about somewhere)
[image]
[image]
[image]
[image]
[image]

nyctomanica, 2024-11-19:
I'm going to have a go with the same-ish config, but a different esp32 board, will report back :)

nyctomanica, 2024-11-19:
That didn't seem to help, although I do now have some new errors...

NO_TRANSITION, and ERROR_PROTOCOL, as well as the original NO_CHANGE_TOO_LONG

[logs_heat-o-matic_logs.txt](https://github.com/user-attachments/files/17817208/logs_heat-o-matic_logs.txt)

Using @olegtarasov's board type, but same config as before.

I'll give another ESP32 board a go now

nyctomanica, 2024-11-19:
Interestingly Tasmota's Opentherm module doesn't seem to work with either board, so I'm leaning towards a shield/boiler issue...

FreeBear-nc, 2024-11-19:
>  A new DIYLESS one is on the way though, so I'll test that once it arrives.

You get a bright green blinky LED to confirm the communication is working. Personally, I find the LED a bit too bright...

nyctomanica, 2024-11-23:
@FreeBear-nc quick question, which version of the Vitodens 050-W do you have? Is it the one with the white front panel (BPJC)?

I've received a new thehognl shield (thought I'd give a new one a go in case it was broken, still waiting for the DIYLESS), but I still have no communication at all with the boiler with the same log messages.

Makes me think that the boiler either isn't OpenTherm compatible, or has some hidden setup menu to enable it which I can't find details about...

Also asked Viessmann about how to connect an OpenTherm stat, but they just told me to do exactly what I've been doing (terminals 1+2 underneath)!

So until the DIYLESS shield (and their ESP32, I ordered the combo ESP+shield thermostat) arrives, I'm at a loss!

FreeBear-nc, 2024-11-23:
> @FreeBear-nc quick question, which version of the Vitodens 050-W do you have? Is it the one with the white front panel (BPJC)?
> Also asked Viessmann about how to connect an OpenTherm stat, but they just told me to do exactly what I've been doing (terminals 1+2 underneath)!

Have a BOKA which is the current version of the 050 (installed last year). Your comment about the connection terminals reminded me that I did have  problem initially with any OpenTherm thermostat not working - The ruddy terminals were wired up the wrong way round. Traced the wiring from the boiler motherboard and found the OT connections were on terminals 3 & 4 ffs....
But to open up the casing to find out requires you to be Gas Safe registered if in the UK (how are they going to find out...).

nyctomanica, 2024-11-23:
Had a nosey at the internal connections, and everything is annoyingly wired up exactly as the installation and service manual says it should be.

Thought I'd try reseating the connectors anyway, put it back together, and got exactly the same NO_CHANGE_TOO_LONG, NO_TRANSITION and ERROR_PROTOCOL errors as before.

I'm now really wondering if there's some kind of menu that the manual doesn't mention to enable OT, as a normal relay on/off stat works as expected...

Either way, I'm stumped at this point

FreeBear-nc, 2024-11-23:
Measure the voltage between terminals 1 & 2 on the boiler. It should be between 18V and 24V. If you get nothing, remove the shorting link on terminals 3 & 4, and measure the voltage across those two.
Beyond that, I'm at a bit of a loss as to what to suggest. Your boiler appears to support OpenTherm, and the specification had been fully developed by the time your boiler was manufactured.

nyctomanica, 2024-11-23:
There's 23v over 1 and 2, but no shorting link on 3 and 4!

I wonder if that's what's doing it!

FreeBear-nc, 2024-11-23:
> There's 23v over 1 and 2, but no shorting link on 3 and 4!
> 
> I wonder if that's what's doing it!

Ah. terminals 3 & 4 should be for an external temperature sensor used for weather compensation. Do you have a fault code of 18 displayed ?
This would be "external sensor open circuit". Try putting a link in between terminals 3 & 4 although that might trigger a 10 fault code.
With no OpenTherm comms, you should also be getting a 5A error code displayed (if I'm reading the manula correctly)

rhtenhove, 2024-11-24:
I've just setup my DIYLESS master shield with a `featheresp32` (an Adafruit Huzzah32 I had lying around).

It works partially, as in, I can control DHW, but not CH. Changing anything about CH doesn't show any response from the boiler. The sensors have worked a few times, showing water and return temperature, as well as relative modulation levels, but are stuck on Unknown now with my current setup. Tells me the boiler is good, it's either the shield or the software at fault.
My logs are filled with one of the previously mentioned errors:

```
[13:25:54][D][opentherm:317]: Sending request with id 0 (STATUS)
[13:25:54][D][opentherm:318]: 00000000 00000000 00000000 00000000
type: READ_DATA; id: 0; HB: 0; LB: 0; uint_16: 0; float: 0.000000
[13:25:54][W][opentherm:353]: Protocol error occured while receiving response: type: NO_CHANGE_TOO_LONG; data: 00000000; clock: 1; capture: 00000000000000000000000111111111; bit_pos: 0
```

It's unclear to me what it's trying to do here, and what the failure message may indicate. Perhaps too slow of a response from the boiler, as in, unexpected response?

Previously I had the `NO_TRANSITION` error, but that doesn't show up anymore. That may be related to having removed a bunch of sensors and switches to rule out any unsupported commands.

My method was hardly scientific, so no clear relation to a specific change I made so far.

I'll keep experimenting and keep track of what's happening after what change.

nyctomanica, 2024-11-25:
Well, it seems something is up with my boiler...

I tried the ESP thermostat with a friend's Ideal boiler, and it worked first time.
Took it back to mine, and the errors returned.

Plus, their Nest would not connect to my boiler either in Opentherm mode...

Bugger. I guess I'll have to get that fixed!

FreeBear-nc, 2024-11-25:
Pester Viessmann tech support and see if they can come up with a low cost (free ?) fix if it is a design flaw with the boiler.

FreeBear-nc, 2024-11-26:
Just a thought - If the BPJC model only supports OT/- (or OpenTherm-Lite), that would explain the lack of communication.
Section 6 of the [OpenTherm Protocol V2.2](https://github.com/FreeBear-nc/opentherm-docs/blob/main/Opentherm%20Protocol%20v2-2.pdf) describes the OT/- application layer. Somewhat crude in that the controller sends a PWM signal to set the boiler temperature, and the boiler responds with one of two message IDs (temperature set point and status).

ESPHome has a PWM component, so it still be possible to have minimal control over the boiler. Using the OpenTherm component to read the status could be a challenge. Ping me an email, and I can help in crafting an esphome.yaml that might work.

olegtarasov, 2025-01-06:
Seems to be a hardware issue with the boiler, closing this issue.

closed 2025-01-06 by olegtarasov