olegtarasov/esphome-opentherm #24: Does not work with GPIO41 as the output pin of esp32s3

ihormelnyk, 2024-11-28:
Hi @olegtarasov 
I have such configuration:
```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

opentherm:
  in_pin: 7
  out_pin: 41
```
The output pin always has high logic level.

I've made some investigation of the opentherm hub.
It does not call out_pin_->setup() function, so the pin is not initialized properly.

Could you please fix this in the next version?
Thanks.

olegtarasov, 2024-11-30:
Hi @ihormelnyk! I've added setup() calls in `develop` branch. You can reference it as an external component:

```yaml
external_components:
  source: github://olegtarasov/esphome-opentherm@develop
  refresh: 0s
```

Please let me know if it solves your issue.

olegtarasov, 2024-12-01:
Yes, protocol errors are common when other components are defined in config. I had zero protocol errors when there was only OpenTherm component defined, but they start popping up as soon as you add something complicated like HA API, displays or temperature sensors. I suspect these components are disabling interrupts from time to time and we lose parts of the message. Anyway, I found that these sporadic protocol errors are harmless, at least to my boiler.

Glad your issue is solved!

closed 2024-12-01 by olegtarasov