olegtarasov/esphome-opentherm #25: Is it possible to use this component for pure boiler monitoring

senilix, 2024-11-30:
First, thank you for getting this component into ESPHome.

I have been been trying to connect an esp32 d1 mini using diyless interface to my Baxi boiler. According to the boiler documentation the boiler supports opentherm although the installation manual specifies opentherm interface V.4. I assume it is backwards compatible with older versions.

As a start I would like to just monitor my boiler operation. Not doing actual thermostat operation. I have made a configuration with only sensor and binary sensor definitions.

```
opentherm:
  in_pin: GPIO21
  out_pin: GPIO22
  #opentherm_version: 2.2 
  
sensor:
  - platform: opentherm
    rel_mod_level:
      name: "Boiler Relative modulation level"
    t_boiler:
      name: "Boiler water temperature"
    t_ret:
      name: "Boiler Return water temperature"

binary_sensor:
  - platform: opentherm
    ch_active:
      name: "Boiler Central Heating active"
    dhw_active:
      name: "Boiler Domestic Hot Water active"
    flame_on:
      name: "Boiler Flame on"
    fault_indication:
      name: "Boiler Fault indication"
      entity_category: diagnostic
    diagnostic_indication:
      name: "Boiler Diagnostic event"
      entity_category: diagnostic

```
It doesn't seem to work. I get receive timeout on "status" request. This made me wonder if it's actually possible to only do monitor operation only?

Sorry if this is not the correct place to ask.

FreeBear-nc, 2024-11-30:
Uncomment the opentherm_version and set it to 4
If the Baxi is anything like my Viessmann, it will expect the control to be setting CH and DHW (if it is a combi) temperatures and ignore any front panel settings. So you will need t_set at a minimum.

FreeBear-nc, 2024-12-01:
Try:
```
opentherm:
  in_pin: GPIO21
  out_pin: GPIO22
  controller_id: 4
```
I found a note from last year from another Baxi user that found the controller ID needed to be 4 or higher.

senilix, 2024-12-01:
Thank your for your help and suggestions.

It seems controller_id isn't a supported and it If i change the opentherm_version to 4  it doesn't seem to make any difference. I get a "[opentherm:357]: Receive response timed out at a protocol level" on the OT_VERSION_CONTROLLER request.

[image]

I tried to add this from the example on the opentherm component documentation.
```

opentherm:
  in_pin: GPIO21
  out_pin: GPIO22
  opentherm_version: 4 
  dhw_enable: true    # Note that when we specify an input in hub config with a static value, it can't be
                      # changed without uploading new firmware. If you want to be able to turn things on or off,
                      # use a switch (see the ch_enable switch below).
                      # Also note that when we define an input as a switch (or use other platform), we don't need
                      # to set it at hub level.

output:
  - platform: opentherm
    t_set:
      id: t_set
      min_value: 20
      max_value: 65
      zero_means_zero: true
      
sensor:
  - platform: opentherm
    rel_mod_level:
      name: "Boiler Relative modulation level"
    t_boiler:
      name: "Boiler water temperature"
    t_ret:
      name: "Boiler Return water temperature"

binary_sensor:
  - platform: opentherm
    ch_active:
      name: "Boiler Central Heating active"
    dhw_active:
      name: "Boiler Domestic Hot Water active"
    flame_on:
      name: "Boiler Flame on"
    fault_indication:
      name: "Boiler Fault indication"
      entity_category: diagnostic
    diagnostic_indication:
      name: "Boiler Diagnostic event"
      entity_category: diagnostic

switch:
  - platform: opentherm
    ch_enable:
      name: "Boiler Central Heating enabled"
      restore_mode: RESTORE_DEFAULT_ON
```
It might indicate that there is a more basic problem. The boiler installation manual doesn't specify that anything needs to be enabled in order for the opentherm port to work.  I measure 24V on the OT wires from the boiler when they are unloaded, so I assume that this is fine.

FreeBear-nc, 2024-12-01:
Make and model of the boiler ?

Details on enabling OpenTherm mode on my boiler (Viessmann 050) consists of a couple of lines, and there is zero information on fault finding. Having looked at installation/service manuals from other manufacturers, this appears to be common.

FreeBear-nc, 2024-12-01:
Try this patch and recompile your code:
```
diff --git a/components/opentherm/schema.py b/components/opentherm/schema.py
index a58de8e..96c1952 100644
--- a/components/opentherm/schema.py
+++ b/components/opentherm/schema.py
@@ -427,21 +427,21 @@ SENSORS: dict[str, SensorSchema] = {
     ),
     "device_id": SensorSchema(
         description="Device ID code",
         unit_of_measurement=UNIT_EMPTY,
         accuracy_decimals=0,
         state_class=STATE_CLASS_NONE,
         disabled_by_default=True,
         message="DEVICE_CONFIG",
         keep_updated=False,
         message_data="u8_lb",
-        order=4,
+        order=0,
     ),
     "otc_hc_ratio_ub": SensorSchema(
         description="OTC heat curve ratio upper bound",
         unit_of_measurement=UNIT_EMPTY,
         accuracy_decimals=0,
         state_class=STATE_CLASS_NONE,
         disabled_by_default=True,
         message="OTC_CURVE_BOUNDS",
         keep_updated=False,
         message_data="u8_hb",
@@ -878,14 +878,14 @@ SETTINGS: dict[str, SettingSchema] = {
         default_value=0,
         order=5,
     ),
     "controller_id": SettingSchema(
         description="Controller ID code",
         message="CONTROLLER_CONFIG",
         keep_updated=False,
         message_data="u8_lb",
         backing_type="uint8_t",
         validation_schema=cv.int_range(min=0, max=255),
-        default_value=0,
-        order=5,
+        default_value=4,
+        order=0,
     ),
 }
```
In your yaml:
```
opentherm:
  in_pin: GPIO21
  out_pin: GPIO22
  controller_id: 4 
  dhw_enable: true
```

Side note: Enabling DHW on some boilers turns off Eco mode and causes the boiler to fire up at regular intervals to keep internals hot. This is so that you get minimal delay in hot water when turning on a tap. Wasteful in terms of gas. On my boiler, still get hot water even with dhw_enable set to false.

senilix, 2024-12-01:
Thanks, just to clarify for my understanding and thank you for your patience.

At the moment I'm using the opentherm component that comes with the latest esphome release in HA. In order add this patch I should either fork the esphome-opentherm component and add the patch and reference the fork using the external_components in  my yaml-file to get the updated component or clone the repo locally on my HA device and reference the path in my yaml-file?

I'm currently reading through the installation manual for my Boiler to see if I can find a hint to how to enable the Opentherm port in case it isn't enabled default.

FreeBear-nc, 2024-12-01:
Clone the esphome-opentherm directory, and then reference it in your yaml with:
```
external_components:
  - source:
      type: local
      path: /<full path >/esphome-opentherm/components
```
You can either apply the patch as posted, or dive in to schema.py and make the changes manually (I had included the extra context if you wanted to go there).

I also have a copy of an older esphome-opentherm repository that has a "fix" rolled in just for Baxi boilers - Would need to do a couple of tweaks to make sure the yaml file is interchangeable.

senilix, 2024-12-02:
Thanks, @FreeBear-nc Your are just awsome.

I just tried out your branch and ran into a few issues. Is it because I try to compile it within my Homeassistant ESPHome instance?

[image]

I phoned the distributor of Baxi boilers her in Denmark today. Unfortunately they didn't know much about opentherm. But, they would try to find out if it has to be enabled to work.

FreeBear-nc, 2024-12-02:
Can you post your yaml in its entirety please.
Just did a compile here, and it runs to completion without issue.

senilix, 2024-12-02:
Sure. Perhaps it's the few small changes I made.
```
esphome:
  name: baxiboiler
  friendly_name: Baxi Boiler

  platformio_options:
    lib_deps:
    - https://github.com/freebear-nc/opentherm_library.git

external_components:
  source:
    type: git
    url: https://github.com/FreeBear-nc/esphome-opentherm.git
    ref: main

esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "wbWkhWzzBW0PC+m5c0s9lCwt2We84U+Eps0ctu9MjWQ="

ota:
  - platform: esphome
    password: "267767d8833db221d890c59095dac327"

wifi:
  ssid: !secret wifi_ssid_sensor
  password: !secret wifi_password_sensor

mdns:

opentherm:
  in_pin: 21 #GPIO21
  out_pin: 22 #GPIO22
  #opentherm_version: 4 
  controller_id: 4
  dhw_enable: true    # Note that when we specify an input in hub config with a static value, it can't be
                      # changed without uploading new firmware. If you want to be able to turn things on or off,
                      # use a switch (see the ch_enable switch below).
                      # Also note that when we define an input as a switch (or use other platform), we don't need
                      # to set it at hub level.

output:
  - platform: opentherm
    t_set:
      id: t_set
      min_value: 20
      max_value: 65
      zero_means_zero: true
      
sensor:
  - platform: opentherm
    device_id:
      name: "Boiler id"

  - platform: opentherm
    rel_mod_level:
      name: "Boiler Relative modulation level"
    t_boiler:
      name: "Boiler water temperature"
    t_ret:
      name: "Boiler Return water temperature"

binary_sensor:
  - platform: opentherm
    ch_active:
      name: "Boiler Central Heating active"
    dhw_active:
      name: "Boiler Domestic Hot Water active"
    flame_on:
      name: "Boiler Flame on"
    fault_indication:
      name: "Boiler Fault indication"
      entity_category: diagnostic
    diagnostic_indication:
      name: "Boiler Diagnostic event"
      entity_category: diagnostic

switch:
  - platform: opentherm
    ch_enable:
      name: "Boiler Central Heating enabled"
      restore_mode: RESTORE_DEFAULT_ON
```

FreeBear-nc, 2024-12-02:
Hrmmm... Went tits up on me too.... However, a minor edit sorted that:
```
sensor:
  - platform: opentherm
    device_id:
      name: "Boiler id"
    rel_mod_level:
      name: "Boiler Relative modulation level"
    t_boiler:
      name: "Boiler water temperature"
    t_ret:
      name: "Boiler Return water temperature"
```

senilix, 2024-12-03:
Ah, my bad. I had the "platform" twice when I copied.

I tried this and it did change things. I still have some issues, but they seem to be of a different character.
[image]
I tried to set "sync_mode: true" just to see if it makes any difference.
[image]
Does change things, but not a fix. I do not have any dallas temperature sensor attached, as you can see from my yaml.

Judging from some googling on other Baxi boilers and opentherm, it does seem that there perhaps is a wire jumper that might need to be removed for it to work. I have to investigate this a bit further.

FreeBear-nc, 2024-12-03:
If I'm translating the installation manual correctly and it is the correct one for your boiler:
https://mitbaxi.dk/Media/Varedokumenter/Installationsvejledning%20WGB-K%20EVO%2020-28%20i.pdf
OpenTherm should be enabled by default, and there are no jumpers that need removing.

If you are stacking the diyless master shield on the ESP32 D1-mini, the in pin should be 19, and the out pin, 23 - But if you are using jumpers, ignore that.

senilix, 2024-12-03:
Yes, it was also my understanding that it should be working default. However, the distributor did mention that it perhaps was a on/off input. I don't know if he mixed it up with some other connection on the boiler. As mentioned, they really didn't seem to know much about opentherm. I do hope they get back to me with some further information. 

I have my ESP32 D1-mini stacked directly on top of the diyless master shield. There are 2 rows of pin headers on each side. The inner pair seem to be directly the same as ESP8266 D1 mini. As far as I can determine it should be IO21 for in and IO22 for out. I can see the slave/boiler LED blinks which I assume is when the ESP32 tries to send commands to the boiler.

I do also have some ESP8266 d1-minis. I could try one of these just to double check that I haven't got pin configuration problem.

senilix, 2024-12-14:
@FreeBear-nc 

Finally some progress. I haven't got any feedback from Baxi. The danish distributor replied to me that they have forwarded my question to Baxi. So perhaps they will reply. 

Anyways, getting back to it today with fresh eyes, I decided to double check that the connection was actually wired correctly to the OT terminals. They were. After that I checked if the OT Channel 1 (prog nmb: 6351) was indeed set to extern room control 1 as specified. This also checked out. Then there was room controller VK1 (prog. nmb: 6355) which was set to "intern". Reading the installation manual it seemed this had some relation to OT so I changed that to "extern". And suddenly the boiler started to answer the commands on the opentherm bus.

So for now I can communicate with the boiler. I havde switch to intern again, since I need to study and prepare some better control for the boiler. But, at least I can now communicate with it.

Thank you for all your help for now. I might get back if I run into trouble. I'm not sure if I need to continue to use your github repo or if I can switch to the officel ESPHome library. If not, I should perhaps fork your repos for now.

FreeBear-nc, 2024-12-14:
> Thank you for all your help for now. I might get back if I run into trouble. I'm not sure if I need to continue to use your github repo or if I can switch to the officel ESPHome library. If not, I should perhaps fork your repos for now.

The official esphome repo is where to find the stable code.

Oleg's esphome-opentherm repo is where new stuff is developed before being pushed to esphome.

My esphome-opentherm repo mirrors Oleg's in the branches (could lag behind), whilst main tries to maintain compatibility using Ihor Melnyk's opentherm library. Its main use would be to check if an opentherm bug is in the core code or in the underlying library.

closed 2025-01-06 by olegtarasov