olegtarasov/esphome-opentherm #26: DATA_INVALID while setting numeric setpoint.

TigroEHOT, 2024-11-30:
Hi there!

Thank you for your hard work and for sharing it with others!

I've got Zota MK-X electric heater and I've tried to interact with the boiler setting the ch_setpoint as number, but noticed that current setpoint doen't change on boiler's panel. Setting it manually works, and boiler tries to maintain the water temperature changing modulation value. I've looked in the logs and found this:

`[20:35:26][D][opentherm:317]: Sending request with id 1 (CH_SETPOINT)
[20:35:26][D][opentherm:318]: 00000001 00000001 00100001 00000000
type: WRITE_DATA; id: 1; HB: 33; LB: 0; uint_16: 8448; float: 33.000000
[20:35:26][D][opentherm:140]: Received OpenTherm response with id 1 (CH_SETPOINT)
[20:35:26][D][opentherm:141]: 00000110 00000001 00000000 00000000
type: DATA_INVALID; id: 1; HB: 0; LB: 0; uint_16: 0; float: 0.000000`

Here I'm trying to set it to 33 degrees. Every time I try to change the setpoint I get this error. 

The config is as follows:

`number:
  - platform: opentherm
    t_set:
      name: "Boiler Control setpoint"
      id: setpoint
      min_value: 20
      max_value: 45
      restore_value: true
      initial_value: 35`

olegtarasov, 2024-12-05:
Hi @TigroEHOT! Sorry for the delay. It seems like your boiler recognizes the message, but refuses to accept the value. Looks like it's a problem with your particular boiler model, so there is not much I can do :(  The only way to debug these kind of issues would be:

1. Get an OpenTherm bridge that can work in a gateway mode (connects both to your boiler and stock thermostat that is known to work correctly). One example is https://github.com/jpraus/arduino-opentherm.
2. Wire up stock thermostat, OT bridge and your boiler. Flash a firmware that can log messages from stock thermostat to the boiler. https://github.com/jpraus/arduino-opentherm has a sketch that does just that.
3. Change some settings on your thermostat and record logs from the bridge.

Using those logs there is a chance to reverse engineer the way your particular boiler operate.

TigroEHOT, 2024-12-08:
Hi there, Oleg!
Reporting: I've tried to change another parameter - max_t_set and there it was! This value affects the water set temperature, it changes the set point on the boilers panel, and boiler maintains that set temperature. So it **is** the particular boiler or manufacturer problem. 
The conclusion:

For Zota electric boilers (maybe for Baxi Ampera also as they are made by Zota for Baxi, but it needs to be checked) to set the water set point instead of t_set use max_t_set value (set as number or output). 

```    
    number:
      max_t_set:
      name: "Max t_set allowed"
      min_value: 20
      max_value: 50
```

Thanks again for you effort and **great** job!
Please, feel free to close the issue.

olegtarasov, 2024-12-08:
@TigroEHOT, great! I'm happy it worked out for your boiler :)  Looks like I need to start a wiki with esphome configurations for different boilers.

closed 2024-12-08 by olegtarasov