olegtarasov/esphome-opentherm #29: Is it possible to implement a Number variable

Csikkovich, 2024-12-20:
Currently when I create an opentherm number, I cannot change how the number should be displayed in the UI (currently only slider). It would be very helpful if it could be changed to box via the mode variable.

FreeBear-nc, 2024-12-21:
Can you post your yaml.
Tried changing the number style from box to slider using mode, and it works for me.
```
number:
  - platform: opentherm
    max_t_set:
      name: "Boiler Max Setpoint"
      id: t_max_sp
      entity_category: config
      min_value: 20.0
      max_value: 75.0
      step: 0.1
      mode: "slider"
```

Csikkovich, 2024-12-21:
My bad, I tried to do it one level out, so on the same level as max_t_set, I didn't even think to try it as you did. Thank you! It is a little confusing but it makes sense seeing yours

closed 2024-12-22 by olegtarasov