olegtarasov/esphome-opentherm #30: Lots of NO_CHANGE_TOO_LONG errors after upgrading to ESPHome 2024.12

olegtarasov, 2025-01-04:
ESPHome 2024.12 upgraded to [ESP-IDF](https://github.com/espressif/esp-idf/) 5.1.5, and lots of `NO_CHANGE_TOO_LONG` warnings started to appear for those who use ESP-IDF in their devices. I didn't have time to debug this issue, but here are some workarounds:

* Check whether your boiler actually works. I found that despite all these warnings many messages get through to the boiler and it works correctly. If that's your case as well, you can just ignore the warnings for now.
* Use Arduino framework instead of IDF in your device configuration. Arduino-based devices seem to not be affected.
* Rollback to ESPHome `2024.11.3` and use OpenTherm component as an external component from this repo:

```yaml
external_components:
  source: github://olegtarasov/esphome-opentherm@main
  refresh: 0s
```

I don't have an estimate for fixing this bug, but I will look into it as soon as I have time.

FreeBear-nc, 2025-02-22:
Done a rebuild today using esphome-2024.12.4 - Getting one every 3-5 minutes NO_CHANGE_TOO_LONG errors. Probably due to a pair of DS18b20 sensors also being polled.
Still running on an ESP-D1 mini (esp8266).

ihatethecloud, 2025-03-21:
I get more of these
```
Protocol error occured while receiving response: NO_TRANSITION
```
alongside NO_CHANGE_TOO_LONG. Might be related?

Boiler is working but starts randomly after 1-2 hours or.

olegtarasov, 2025-10-18:
Low-level boiler communication on ESP32 was completely rewritten using RMT. Current dev version is here: https://github.com/olegtarasov/esphome/tree/opentherm-dev. I will merge a more or less stable version of the code to this repo in a couple of days. Protocol errors are not completely gone, but I suspect the actual boilers are to blame in some cases.